riscv/riscv-fast-interrupt #120: Incorrect spec of behavior on WFI (Section 6.3)

kasanovic, 2021-03-02:
A higher-mode threshold setting can cause lower-mode interrupts that are enabled and above the lower-mode threshold to not wake up a WFI in the higher-mode. When a currently pending higher-mode interrupt is below the threshold set in the higher mode, the higher-mode interrupt will effectively mask the lower-mode interrupt in the interrupt priority reduction tree, but will not cause WFI to exit in the higher-mode due to the higher-mode threshold setting.  Current text does not capture this behavior correctly.

When the higher-mode threshold is set to zero, it cannot mask any pending and enabled lower-mode interrupts.

dansmathers, 2021-03-04:
added pull request #121.  please review.  thanks!

dansmathers, 2021-04-13:
can this issue be closed with pull #144?

Kevin-Andes, 2021-04-15:
I thought Krste mentioned (in our group meeting on 2021/04/13) that he had some trouble parsing it and suggested a little cleanup...

For example, why is the interrupt level a function of clicintie[__i__]? Do you mean clicintctl instead?  Also, why would the interrupt level be 0? Doesn't 0 implies "no interrupt"?  Maybe you can further elaborate a little bit to make it easier for first-time readers to understand.

dansmathers, 2021-04-15:
Sorry. I closed pulls #135 and #121.  can you instead review the pull #144?  

> why would the interrupt level be 0?
if clicintctrl has 8 implemented bits, it is possible for clicintctrl to be set to 0.  but since bit 0 implies "no interrupt" no interrupt will be taken and WFI will not wake up, but it will block lower privilege pending interrupts in the priority encoder, so it is a corner case that needs to be considered.    
See issue #104:
if there is a non-zero level user interrupt pending and a level-zero machine interrupt pending, the machine interrupt will be the global maximum across all pending-and-enabled interrupts but will not cause an interrupt to be taken according to the table. so a level-zero pending machine interrupt will mask any lower privilege interrupts.

I updated pull #144 to add this note.  We can wait until discussion in our next TG before approving this pull.