riscv/riscv-fast-interrupt #122: Wording in section 3.2.1

konrad-schwarz, 2021-03-04:
I found the wording below confusing:

> The 2-bit cliccfg.nmbits WARL field specifies how many bits of actual registers are implemented in
> clicintattr[i].mode to represent an input i's privilege mode. Although cliccfg.nmbits field is always 2-bit wide,
> the actual register bits implemented in this field can be fewer than two (depending how many interrupt privilege-modes
> are supported).

I originally thought the phrase "how many bits of actual registers are implemented" referred to addressing bits or something.  How about: "[nmbits] specifies the number of writable bits in the clintattr registers' mode field.  The mode field is defined to be two bits wide, but can contain zero, one, or two writeable bits."

konrad-schwarz, 2021-03-04:
I find the next paragraph confusingly worded as well.

> For example, in M-mode-only systems, only M-mode exists so we do not need any extra bit to represent the supported
> privilege-modes. In this case, there is no actual registers needed/implemented in the clicintattr.mode and thus
> cliccfg.nmbits is 0 (i.e., cliccfg.nmbits can be hardwired to 0 without using any register either).

"For example, in M-mode-only systems, interrupts can only be taken to M-mode, so the mode is implicitly clear; clicintattr.mode can be hardwired to zero."

Also, I think you are using the word "register" in the Verilog sense, as a kind of variable; whereas in ISA/computer architecture parlance, a register is a (normally) word-wide location that is accessible to software as a whole.  E.g., "memory-mapped register".

Kevin-Andes, 2021-03-04:
Thank you for your great suggestions and I will revise the spec accordingly. 

By the way, if we use the word "flip-flops" instead of "registers", would that sound better and clearer? Or is that also confusing? What is a better way to describe the physical storage bits?

Thanks.

konrad-schwarz, 2021-03-04:
Well, I think your "physical storage bits" isn't too shabby.  E.g., "The field [...] describes how many bits are physically implemented in [...]."

(I'd avoid "flip-flops" as too EE jargon and maybe there are technological alternatives to using flip-flops (e.g., an emulated system); I don’t recall ever seeing that word in documentation of this kind).

Kevin-Andes, 2021-03-04:
Thank you for the great advice!

closed 2021-03-30 by Kevin-Andes