riscv/riscv-fast-interrupt #125: Typo in definition of clicintattr[i].mode?

nati-rapaport, 2021-03-22:
clic spec, chapter 3.2.1 (Specifying Interrupt Privilege Mode) specifies: "Similarly, in systems that support all M/S/U-mode interrupts, cliccfg.nmbits can be set to 0, 1, or 2 bits to represent privilege-modes. cliccfg.nmbits = 0 indicates that all local interrupts are taken in M-mode. cliccfg.nmbits = 1 indicates that the MSB selects between M-mode (1) and U-mode (0)." etc.
On the other hand, the table below shows that when nmbits=1 and clicintattr[i].mode=0x, then the interpretation is S mode interrupt (see below). There is a contradiction here. So, who is right here - the wording or the table?

priv-modes  nmbits clicintattr[i].mode Interpretation
M                    0              xx                    M-mode interrupt
M/U                0              xx                    M-mode interrupt
M/U                1              0x                    U-mode interrupt
M/U                1              1x                    M-mode interrupt
M/S/U             0              xx                    M-mode interrupt
**M/S/U             1              0x                    S-mode interrupt**

dansmathers, 2021-03-22:
This looks like the same as issue #117.

Kevin-Andes, 2021-03-23:
Yes, this is an inconsistency and was also reported earlier in issue #117 as Dan mentioned. Sorry we did not have time to discuss this issue in our last week's meeting. We will try to discuss and resolve this in our next week's group meeting.

Thanks.

kasanovic, 2021-04-13:
Closed by pull #132

closed 2021-04-13 by kasanovic