riscv/riscv-fast-interrupt #149: 32-bit writes to memory-mapped CLIC registers

JamesKenneyImperas, 2021-04-15:
Can you please confirm whether it is legal to perform a 32-bit write to a memory-mapped block controlling a single interrupt and, if so, whether there is a defined order in which changes take effect?

As a concrete example, suppose that interrupt 0 is currently configured as a level-sensitive interrupt, and the location at offset 0x1000 is written with the 32-bit value 0xffc40101. This simultaneously attempts to reconfigure the interrupt as edge-sensitive and set it pending. This will only work as expected if the change in configuration from level to edge happens _before_ the attempt to set pending (otherwise, for level sensitive interrupts, changes to pending state by software writes are ignored). If this is legal, is the pending bit set or not?

Thanks.

JamesKenneyImperas, 2021-04-24:
I have a related question about the `clicintip[i]` register. The specification says:

_For edge-triggered interrupts, software writes will set/clear pending bits. In contrast, for level-triggered interrupts, software writes to pending bits are ignored completely_

What happens in this case:
1. Software configures an interrupt to be edge-triggered by writing `clicintattr[i]`.
2. Software sets the interrupt pending using `clicintip[i]`.
3. Software reconfigures the interrupt as level-triggered by writing `clicintattr[i]`.

After step 3, is the interrupt still pending, or not? What value is seen if `clicintip[i]` is read?

Thanks.

kasanovic, 2021-04-27:
A 32-bit write is legal.  However, there is no specified order in which the effects of the individual byte updates take effect.

kasanovic, 2021-04-27:
Moving the related question, which is a separate concern to a new issue.

kasanovic, 2021-06-22:
Clarified in https://github.com/riscv/riscv-fast-interrupt/commit/f838d19302060576c0e35229d749369bb71771a9

closed 2021-06-22 by kasanovic