riscv/riscv-fast-interrupt #154: xtval and xepc when access exception on the table load (section 5.4)

JamesKenneyImperas, 2021-04-28:
I don't understand the intended behavior for `xtval` and `xepc` as described in section 5.4. The specification says:

> If there is an access exception on the table load, both `xtval` and `xepc` holds the faulting address.

The subsequent clarification note then says:

> The interesting case is vertical traps, where a more privileged layer is handling page faults or other synchronous faults on vector table access. The regular code path in more privileged layer will want to use `xtval` to determine what bad virtual address to page in, but will not normally restore `xtval` when returning to faulting context (potentially after some time and other contexts have run) However, it will restore `xepc` (using `x` for more privileged mode here) before using `xret` on normal code path. This is a rationale for why both `xtval` and `xepc` are written with the faulting address.

If I read this correctly, this is saying that a return from a vertical trap will execute at an address in the trap vector table. This doesn't seem to make sense, because the trap vector table doesn't contain _instructions_; it holds _addresses_. There is another comment that:

> The `inhv` bit has no effect except when returning from a trap using an `xret` instruction.

However, I can't find any description of this behavioral effect - can you please say where this is described?

Thanks.

kasanovic, 2021-05-11:
The inhv bit when set at xRET informs hardware to repeat the table load using the address in xEPC to obtain the address of the trap handler that is then written to the PC instead of directly writing xEPC to the PC. i.e.,
```
if (inhv)
    PC = M[xEPC]
else
    PC = xEPC
```
One of the goals of this behavior is to avoid complicating the critical code paths for handling virtual memory in the more-privileged layer.  The more-privileged layer does not have to distinguish CLIC vector table reads from other forms of data page fault and can handle them using exactly the same code.

JamesKenneyImperas, 2021-05-12:
Hi Krste,

I understand the intent now.

Presumably the `inhv` bit is also cleared if and only if the `M[xEPC]` load is successful? The new pseudocode doesn't mention when this happens.

Thanks.

JamesKenneyImperas, 2021-05-12:
Also if the `M[xEPC]` load in an `xRET` is _not_ successful, what is the effect on processor state after the `xRET` instruction completes? An `xRET` would normally update various state bits (in `xstatus`, `xintstatus` and `tcontrol`, for example). Do any of these updates occur if the load fails?

JamesKenneyImperas, 2021-05-12:
Sorry, another question: presumably the memory access `M[xEPC]` is performed with _target mode_ access privileges, not _current mode_ access privileges? Could you please update the specification to clarify this?

Thanks.

kasanovic, 2021-05-12:
Spec already says:

Re: Clearing inhv: "If the fetch is successful, the processor clears the low bit of the handler address, sets the PC to this handler address, then clears the xinhv bit in xcause."

also

Re: M[xEPC] load failing on retry - just the same as the first time.  This should be implied by rest of spec.
 
also 

Re: privileges: "For permissions-checking purposes, the memory access to retrieve the function pointer for vectoring is treated as a load with the privilege mode and interrupt level of the interrupt handler. If there is an access exception on the table load, both xtval and xepc holds the faulting address."

dansmathers, 2021-06-22:
just walking though the scenario:
user interrupt 16 starts hardware vectoring, hw sets ucause.uinhv
during table vector read, supervisor page fault occurs.  .scause.sinhv isn't set, correct only ucause.uinhv? 
hw sets stval and sepc to the faulting address 
Since user hw vectoring didn't complete, the user interrupt 16 is still pending and enabled.
supervisor mode handles page fault, fault handler finishes with a sret instruction.
sret will enable user interrupts and switches priv to user mode.  Since user interrupt 16 is still pending and enabled the interrupt will occur before this hardware can take effect, right?
(is the u and s correct?)
SRET instruction:
if (uinhv)
    pc := M[sepc]
else
    pc := xepc

Do we even need that special hardware since it will be interrupted by the pending interrupt which will attempt a hardware vector again?

JamesKenneyImperas, 2021-06-22:
I'm not sure this is quite right. When the exception is taken, the inhv field is set in xcause (i.e. target mode cause) which in this case would be scause.inhv. On exception return, the sequence is:
1. get current mode inhv (i.e. scause.inhv)
2. switch to target mode (User)
3. set pc to M[sepc], with the memory access done in target mode (User)
4. take interrupt which is now pending and enabled.
(This might be moot since N-extension is not being progressed, I think.)

kasanovic, 2021-06-22:
Dan's example had interrupt in user mode, so ucause.uinhv was set.  (Same scenario can occur in S versus HS interrupts, so not moot in general regardless of N-extension status)
S-mode trap handler does not need to know about uinhv, only that it must be restored (or left alone in ucause register).
In this example, the uie bit would be cleared before the table read (hardware commits to take this interrupt before table read starts).  On sret,  uie is restored (to it's original cleared state) so no other interrupt is taken on return.  Assuming table read succeeds this time, the user-mode handler will start running at the address returned by the table read.

kasanovic, 2021-06-22:
Will add text to clarify that xie is cleared before table read.  i.e., when interrupt is being taken, the mode is set to that of the handler x, the xie bit is cleared, then the table load is performed with the permissions corresponding to the handler's mode x with the xcause.xinhv bit set.

JamesKenneyImperas, 2021-06-23:
Hello Krste,

I realise now that there is a subtlety here that I have completely missed: I assumed that the `inhv` bit would be set in the exception _target mode_ `cause` register (because all other recorded state is always in the target mode exception management registers) but you are saying that this one bit of state is put in the _source mode_ `cause` register instead. I think it would be worth adding an explicit note explaining this, because it is not what I expected. Is there a particular reason that it has do be done like this? It would seem to complicate exception management if the handler ever needs to discover whether it was activated by a trap vector table fetch exception.

Thanks.

commit dd15cd3c (2021-05-11): Added explanation of xinhv operation during xret.
Closes #154.