riscv/riscv-fast-interrupt #155: Clarify clicinttrig details

JamesKenneyImperas, 2021-04-28:
Section 4.8 says that the detailed behavior of interrupt triggers is defined in the Debug Specification. I have looked in that specification and can see no clear way that the corresponding state there (tmexttrigger?) ties up with the memory-mapped state here. Can you please clarify the intended use model?

Thanks.

dansmathers, 2021-06-08:
clarify inttrig details.  Entire meeting discussed CLIC trigger details with Tim from debug group.   CLIC, trigger module, and hart need to be tightly integrated since these features need to be in the core to get sync behavior.  discussion on where the details are written down.  E.g. CLIC spec describes set of interrupts selected for interface to debug, debug specifies what happens on that event (enter debug mode, start trace, stop trace).  For now, keeping abstract action in CLIC, debug takes concrete action.  try do discuss more over email reflector.

kasanovic, 2021-06-08:
The CLIC clicinttrig registers specify how and when triggers are sent to the debug hardware.  The debug hardware registers specify how they handle those generated triggers.

dansmathers, 2022-06-09:
closed with pull #190.

closed 2022-06-09 by dansmathers