riscv/riscv-fast-interrupt #156: Clarify isolation mechanism over CLIC memory mapped registers

joxie, 2021-05-25:
Per discussions in fast interrupt TG we are expecting memory mapped registers to be aliased in M/S mode systems, it would be good to provide some clarity in spec to avoid confusion.

For example, below statements may make people think that CLIC is enforcing priv mode identifier on memory bus when accessing these registers:

All CLIC-memory mapped registers are visible to M-mode. Interrupt registers clicintip[i], clicintie[i], clicintattr[i], clicintctrl[i] configured as M-mode interrupts are not acessible to S-mode and U-mode. Interrupt registers clicintip[i], clicintie[i], clicintattr[i], clicintctrl[i] configured as S-mode interrupts are not acessible to U-mode.

In S-mode, any interrupt i that is not accessible to S-mode appears as hard-wired zeros in clicintip[i], clicintie[i], clicintattr[i], and clicintctl[i].

Another example is that cliccfg might be RO for S mode and RW for M mode, but memory map says it is RW:

  0x0000         1B          RW        cliccfg

kasanovic, 2021-06-08:
Will add clarification to spec about how memory apertures are intended to be protected across privilege modes.

kasanovic, 2021-06-08:
In TG meeting 6/8/2021 we realized the discussion of memory apertures was incorrectly removed from the spec, and need to be reinstated to remove this confusion.

kasanovic, 2021-06-22:
Added back text on memory apertures in commit https://github.com/riscv/riscv-fast-interrupt/commit/f2c9a1e7ef6fab9eb69bff39b0c0f1d677dc4561 .  Still needs a little clean up to help clarify these issues.

commit 935c046c (2021-07-06): Clarified text on memory map for different privilege modes.
Closes #156.
Closes #77.
Closes #79.
closed 2021-07-06 by kasanovic