riscv/riscv-fast-interrupt #185: SAIL implementation of CLIC

dansmathers, 2021-12-21:
SAIL implementation needed for Definition of Done.  Creating an issue for tracking purposes.

dansmathers, 2022-01-09:
Replace pseudo-code in spec with SAIL code once SAIL code is implemented?

dansmathers, 2024-02-10:
FYI, smrnmi may be a good reference for how to implement clic in SAIL
https://github.com/riscv/sail-riscv/pull/396

dansmathers, 2024-02-10:
another reference is an attempt at plic: https://github.com/saravanakumarsastra/sail-riscv

dansmathers, 2024-02-10:
if CLIC changes from mem-mapped clicint registers to using indirect csrs: https://github.com/adlr/sail-riscv/tree/smcsrindpr

dansmathers, 2024-03-01:
Initial sail-riscv clic implementation.   Moving issue to post 1.0 - ecosystem phase as we work to get the pull approved.

https://github.com/riscv/sail-riscv/pull/420
Tested against Tests: https://github.com/riscv-non-isa/riscv-arch-test/pull/436

Note: pulls are not yet available for spike that support CLIC

Related pulls to run arch-tests against the sail-riscv model:
https://github.com/riscv-software-src/riscv-config/pull/169
https://github.com/riscv-software-src/riscof/pull/106
https://github.com/riscv-software-src/riscv-isa-sim/pull/1596

To include m-mode CLIC interrupt tests in riscof testlist flow, add Smclic to riscof yaml file, e.g.:
spike/spike_isa.yaml:
ISA: RV32IMCZicsr_Zifencei_Smclic

christian-herber-nxp, 2025-05-05:
will be tracked in Jira

closed 2025-05-05 by christian-herber-nxp