riscv/riscv-fast-interrupt #193: Motivation of changes for xret instruction

divan0v, 2022-01-19:
Add, please, motivation of changes for 'xret' instruction behaviour.

Motivation for the hardware vectoring bit 'xinhv' is presented in section 5.4: 
> The hardware vectoring bit xinhv is provided to allow resumable traps on fetches to the trap vector table.

Could you explain and add similar note for 'xret' modification. From my opinion, exception on 
hardware vectoring operation is rare case and software could process such exception with old 'xret' 
behaviour manually to allow resumable trap without performance degradation. 

Thanks.

divan0v, 2022-01-19:
Thanks for commit creation. 

The following phrase describes privilege mode for operation M[TBASE + XLEN/8 * exccode]:
> For permissions-checking purposes, the memory access to retrieve the function pointer for vectoring is treated as
> a load with the privilege mode (also obeying MPRV and SUM bits) and interrupt level of the interrupt handler.

Could you also add phrase about privilege mode and interrupt level that is used for operation M[xepc]? Or phrase above covers both cases?

kasanovic, 2022-02-01:
Closed with #193 and #194.

closed 2022-02-01 by kasanovic