riscv/riscv-fast-interrupt #195: Edge case behavior related to xintthresh and xstatus.xie

tovine, 2022-01-27:
There is one edge case that I couldn't find documented in neither the CLIC spec nor existing issues:

In a scenario where an individual interrupt on a low level (e.g. 1) is enabled, but interrupts are globally disabled via the `xstatus.xie` bit:

0. `xintthresh` is 0
1. Low level (e.g. 1) interrupt is raised from the CLIC, but not handled by the hart because `xstatus.xie` is 0
2. `xintthresh` is changed to a higher level (e.g. 2)
3. `xstatus.xie` is now written to 1, enabling interrupts globally (for that privilege level)

What is the expected result after this?

On one hand, the CLIC could lower the interrupt request after event 2, because the interrupt level (1) is now below the threshold (2), but on the other hand, the interrupt has already been presented to the hart - although not taken because it was blocked by `xstatus.xie == 0`.  
On the other hand one could say that since the interrupt is already presented to the hart (which at the time was was a valid action since `xintthresh` was lower than the interrupt's level), and that it _is_ the next interrupt in line to be serviced by the hart (unless other, higher level interrupts have arrived in the meantime).

What do you think? From the current formulation of the spec I can't see a definitive answer to this edge case, so if nothing else a small note clarifying this would be helpful.

dansmathers, 2022-01-27:
I think that case is covered in this table (currently section 8.1)

interrupt enable `ie` = ({status}.{ie} & `clicintie[__i__]`)  and 
interrupt level `L` = max({intstatus}.{il}, {intthresh}.`th`);

 Current  |      CLIC                     |->      Current'          Previous
 p/ie/il    | priv level   id             |->    p/ie/il  pc  cde   pp/il/ie epc
 P  ?  ?    | nP<P     ?      ?          |->    - -  -   -   -     -  -  -  -   # Interrupt ignored  (row1)
 P  0  ?    | nP=P     ?      ?          |->    - -  -   -   -     -  -  -  -   # Interrupts disabled (row2)
 P  1  ?    | nP=P     0      ?          |->    - -  -   -   -     -  -  -  -   # No interrupt           (row3)
 P  1  L    | nP=P   0<nL<=L  ?  |->    - -  -   -   -     -  -  -  -   # Interrupt ignored    (row4)
 P  1  L    | nP=P   L<nL    id      |->    P 0  nL  V   id    P  L  1  pc  # Horizontal interrupt taken (row5)
 P  ?  ?    | nP>P     0      ?          |->    - -  -   -   -     -  -  -  -   # No interrupt (row6)
 P  e  L    | nP>P   0<nL    id      |->   nP 0  nL  V   id    P  L  e  pc  # Vertical interrupt taken (row7)

for your example steps 0,1,2 - when xtatus.xie is 0, that is row1, interrupt ignored.
after your step 3, xstatus.xie is written to 1 but nL<=L, that is row 4, interrupt ignored.

I think that is also the intended behavior of xintthresh where an interrupt handler can raise its priority level to not be interrupted by another interrupt with priority level below xintthresh.

Let me know if you still think we should add a clarification to the xintthresh section of the spec.

tovine, 2022-01-27:
Hi and thanks for the quick and detailed answer!

Based on that table alone it's obvious that this would be the case, but with this text in 8.3 regarding WFI responses it became a bit less obvious again:

> In CLIC, similar to original mode, Wait for Interrupt instruction (WFI) is unaffected by the global interrupt-enable bits in xstatus.xie but should honor clicintie[i] and xintthresh

This makes it seem like when an interrupt is enabled individually and `level >= xintthresh` (or is it strictly `> xintthresh` regardless of whether or not an existing interrupt is being serviced?), it should be presented to the hart regardless of the `xstatus.xie` bit - just not serviced until the `xie` bit is set.
Or is this only for WFI purposes, and not related to actual interrupt handling at all?

tovine, 2022-02-03:
Follow-up question: how should `mintthresh` react regarding interrupt levels that aren't implemented (i.e. bits hardwired to 1)?

Should all writes with a threshold value less than the minimum level (i.e 127 if one bit is implemented, 63 if two bits etc.) either be:
a) Truncated to 0 or
b) Be treated like with level configuration, so that _only_ a 0 write to `mintthresh` actually means 0, and any level between 0 and the minimum supported level wil be rounded up to that minimum level?

dansmathers, 2022-02-03:
Hi,
I think the way the spec is currently written, xintthresh/xintstatus are always implemented as 8 bits.

As you point out, when clicintctl bits aren't implemented, the level is still translated to look like an 8-bit number 
section 4.2.2:
non-existent bits for level encoding, assumed to be 1
CLICINTCTLBITS nlbits clicintctl[i] interrupt levels
      0         2      ........     255
      1         2      l.......     127,255
      2         2      ll......     63,127,191,255
      3         3      lll.....     31,63,95,127,159,191,223,255
      4         1      lppp....     127,255

so the interrupt levels can always be compared against a 8-bit xintthresh/xintstatus value without having to specify truncating or rounding of xintthresh/xintstatus.  

I think you are suggesting a way to save implementing a couple bits of xintthresh?  Maybe we could add to the spec that  xintthresh is WARL to allow for that optimization?  We can discuss in the next TG meeting.  but I think we would still only specify that 
L = max(xintstatus.xil, xintthresh.th);
 P  1  L  | nP=P   0<nL<=L  ?  |->    - -  -   -   -     -  -  -  -   # Interrupt ignored
 P  1  L  | nP=P   L<nL    id  |->    P 0  nL  V   id    P  L  1  pc  # Horizontal interrupt taken

and leave it up to implementations to choose the xintthresh values that can be programmed.
Did I understand your question?

tovine, 2022-02-03:
> I think you are suggesting a way to save implementing a couple bits of xintthresh? Maybe we could add to the spec that xintthresh is WARL to allow for that optimization? We can discuss in the next [...]

Yes, exactly!
From a functional point of view, there's no point in keeping more threshold level bits than you have implemented actual level bits for (apart from maybe _one extra_ to say if the lower bits are to be treated as zero or one).

Making it WARL sounds like a good solution.

tovine, 2022-02-15:
Why would it need to be discovered? Wouldn't it be enough to just look at `CLICINTCTLBITS` and add one (to indicate if threshold is 0 or the minimum configurable interrupt level?

dansmathers, 2022-02-15:
> Wouldn't it be enough to just look at CLICINTCTLBITS and add one (to indicate if threshold is 0 or the minimum configurable interrupt level?

Right.  that's different than the pull request I created that just added WARL to the spec.   The TG was asking for a more limited set of possibilities than WARL.  WARL includes only implementing the even bits, or only the odd bits or implementing no bits.  They agree the spec needs to be updated to allow implementing a minimum number of bits.

dansmathers, 2022-06-09:
closed with pull #217

closed 2022-06-09 by dansmathers