riscv/riscv-fast-interrupt #197: xret and intstatus.xil in section 8.5 return from handlers

dansmathers, 2022-01-31:
I have a question about xret and intstatus.xil in section 8.5 return from handlers

This line in the clic spec says that intstatus.xil will be updated after a xret instruction.

The regular xret instructions are used to return from handlers in privilege mode x. Execution continues at the saved privilege mode xcause.xpp, at PC xepc, with interrupt level xcause.xpil

In the privilege spec, it looks like m-mode can choose to use sret or uret as a shortcut to change modes from m-mode to s-mode or u-mode: 
An xRET
instruction can be executed in privilege mode x or higher, where executing a lower-privilege xRET
instruction will pop the relevant lower-privilege interrupt enable and privilege mode stack. In
addition to manipulating the privilege stack as described in Section 3.1.6.1, xRET sets the pc to
the value stored in the xepc register.

It seems clear from the text that if the hart is in m-mode and it executes a mret, intstatus.mil should be updated.  
As a corner case, if the hart is in m-mode and it executes a sret or uret, should intstatus.mil also be updated?   I think the answer is yes.  If so, should we add text to the clic spec to clarify for when a lower-privilege xRET instruction is used?

dansmathers, 2022-01-31:
Also curious how doing a sret while in m-mode while xcause.inhv of the previous privilege mode is set.   Would the sret execution while in m-mode also modify the xepc behavior?  I would think not but want to check.

When returning from an xret instruction, the xinhv bit modifies behavior as follows:

if (xcause.inhv) //xcause here refers to the cause CSR of the previous privilege mode
    pc := M[xepc]
else
    pc := xepc

aswaterman, 2022-01-31:
@dansmathers for the baseline privileged architecture, `sret` does the same thing in M-mode as it does in S-mode: i.e. it still jumps to the address in `sepc` and enters the privilege mode indicated by `SPP`.

kasanovic, 2022-02-01:
As @aswaterman indicated, xret instructions perform the same operation regardless of the mode in they are executed.  This also means that inhv-related and intstatus-related actions are the same regardless of which mode the xret is executed in.

kasanovic, 2022-07-05:
Close this issue, but carry forward missing hypervisor behavior to issue #247