riscv/riscv-fast-interrupt #202: 64-bit writes to {clicintctl,clicintattr,clicintie,clicintip} text clarification

dansmathers, 2022-02-09:
The current text says:
A 32-bit write to {clicintctl,clicintattr,clicintie,clicintip} is legal. However, there is no specified order in which the effects of the individual byte updates take effect.

I assume the same is true for 64-bit writes (RV64 SD instruction)

Should the text be changed to:
Larger than 8-bit writes to  {clicintctl,clicintattr,clicintie,clicintip} are legal. However, there is no specified order in which the effects of the individual byte updates take effect.

kasanovic, 2022-03-15:
The memory map should require each memory region to begin on a naturally aligned power-of-2 boundary, e.g., 4KiB.
For fields that are greater than 8b, e.g., interrupt number in clicinttrig, the whole field should be updated atomically to avoid partial writes with unexpected behaviors.
To achieve this goal, it might be simpler to specify that 8b, 16, and 32b stores are atomic, as they will be naturally aligned and hence most I/O buses should support at least this granularity of update (in current spec there is no field straddling more than 16b so in practice implementation could be legal with 16b updates).  We can say that 64b stores are handled 32b at a time in some order.
Also, should specify that 64b load accesses can be broken into two 32b loads in any order.
Similarly for the load and store portions of an AMO.

dansmathers, 2022-08-16:
closed by pull #238

closed 2022-08-16 by dansmathers