riscv/riscv-fast-interrupt #208: fence.i required after modification of the trap vector table?

dansmathers, 2022-03-05:
Should CLIC also specify fence.i is required after modification of the trap vector table like what is specified in the Zc proposed extension?

Zc extension - Table Jump Fault handling section:
"Memory writes to the jump vector table require an instruction barrier (fence.i) to guarantee that they are visible
to the instruction fetch."

Created pull #206.

Silabs-ArjanB, 2022-04-11:
The note added in https://github.com/riscv/riscv-fast-interrupt/pull/206/files#diff-3eb538450d2c1f1119dfc066c0ed20f286527c0e4f995a9196c77dfd30b8f7b4R1072 states that a FENCE.I can be used to address the coherency issue that can exist between a store to the trap vector table and the load to retrieve the function pointer from the vector table. 

This does not sound correct as "the memory access to retrieve the function pointer for vectoring is treated as a **load**" whereas a FENCE.I instruction "provides explicit synchronization between writes to instruction memory and **instruction fetches** on the same hart".

Had the CLIC pointer retrieval been handled as an instruction fetch (as done for the table jump in Zc, see https://github.com/riscv/riscv-fast-interrupt/issues/191), then a fence.i would work.

It should not matter that an implementation uses the fetch stage/infrastructure/interface to perform the pointer fetch related memory access, it is still a **load**. (We have now implemented this in the CV32E40X and handling this pointer access as a load is a bit awkward).

Instead of recommending FENCE.I maybe a regular FENCE can be recommended?

dansmathers, 2022-04-26:
closed with pull #227

closed 2022-04-26 by dansmathers