riscv/riscv-fast-interrupt #213: Does CLICMTVECALIGN requirements align with AIA IMSIC and code size extension jump table?

dansmathers, 2022-03-06:
From Allen Baum email:
CLICMTVECALIGN 6-13
The spec says the entire address field can be legal, but allows restrictions like this. This is not complete, though,  as it is legal to require that upper bits be within some range as well  (so someone doesn't set xtvec to an IO location, for example, or forces it to be a ROM location for security or local SRAM only.
That's another thing riscv-config can be configured to explain, for what it's worth)
so, while this sets up a requirement, it isn't really complete, and someone could implement a 64K NAPOT page, for example.  
It would be interesting, by the way, to compare what this does compared to AIA's IMSIC.
It would be great if there was a common core. The big difference is vectored interrupts (code size extension is implementing something like that for jumps)
but a lot of the functionality seems pretty close, but all the formatting of mmio space is different.

dansmathers, 2022-03-06:
code-size extension alignment:
JVT.base[XLEN-1:6] can be programmed. 
Using JVT.base[5:0] is implicitly zero, and is naturally aligned for all legal values of XLEN.

AIA IMSIC:
For convenient addressing, the memory pages of all machine-level interrupt files (or all those of a single
group of harts) should be arranged so that the address of the machine-level interrupt file for hart
number h is given by the formula A + h × 2C for some integer constants A and C. If the largest
hart number is hmax, let k = .log2(hmax + 1)., the number of bits needed to represent any hart
number. Then the base address A should be aligned to a 2k+C address boundary, so A + h × 2C
always equals A | (h × 2C), where the vertical bar (|) represents bitwise logical OR.
The smallest that C can be is 12, with 2C being the size of one 4-KiB page. If C > 12, the start of
the memory page for each machine-level interrupt file is aligned not just to a 4-KiB page but to a
stricter 2C address boundary. Within the 2k+C-size address range A through A + 2k+C - 1, every
20 RISC-V Advanced Interrupt Architecture V0.2.1-draft
4-KiB page that is not occupied by a machine-level interrupt file should be filled with 32-bit words
of read-only zeros, such that any read of an aligned word returns zero and any write to an aligned
word is ignored.
The memory pages of all supervisor-level interrupt files (or all those of a single group of harts) should
similarly be arranged so that the address of the supervisor-level interrupt file for hart number h is
B + h × 2D for some integer constants B and D, with the base address B being aligned to a 2k+D
address boundary.

Should we just change the parameter comment to CLICMTVECALIGN >= 6?

kasanovic, 2022-06-21:
Closed by https://github.com/riscv/riscv-fast-interrupt/pull/216

closed 2022-06-21 by kasanovic