riscv/riscv-fast-interrupt #225: Only require that effects of stores to clicintie/clicintip are evaluated within bounded time (not requiring immediate effects)

Silabs-ArjanB, 2022-04-11:
RISC-V Privileged specification 1.12 states the following about the MIP and MIE csrs:

> These conditions for an interrupt trap to occur must be evaluated in a bounded amount of time
> from when an interrupt becomes, or ceases to be, pending in mip, and must also be evaluated
> immediately following the execution of an xRET instruction or an explicit write to a CSR on which
> these interrupt trap conditions expressly depend (including mip, mie, mstatus, and mideleg).

Can a similar statement(s) be added in the CLIC spec related to clicintie and clicintip? I am particularly looking for a statement saying that determination of whether an interrupt is 'pending and enabled' must be evaluated in a bounded amount of time and that there is **no** requirement that an explicit (store) write to clicintie or clicintip are evaluated immediately.

Note that for CSR write (as for MIP/MIE) it is relatively easy to guarantee interrupt traps on the instruction immediately following such a CSR write. If similar requirements would exist on stores to clicintie/clicintip, then this would seriously complicate designs. The core normally would not even know that a write is to clicintie/clicintip. If a requirement for 'immediate evaluation' would exist, then it would also seriously lower performance as loads and stores can likely no longer be performed in a back-to-back manner when using protocols as AHB. It would also complicate for example pipelining the arbitration tree logic.

kasanovic, 2022-04-26:
Agree.

kasanovic, 2022-05-10:
Change  write-> store to be clearer bound is relative to an instruction execution in same program stream.

closed 2022-05-10 by kasanovic