riscv/riscv-fast-interrupt #230: explicit reference that xepc set to table load address on trap while xcause.xinhv is set?

dansmathers, 2022-04-27:
Peter Ashenden email 4/26/2022:
Suppose the processor gets an S-mode vectored interrupt, and so sets
scause.sinhv and starts a vector table load. Is the intention that the
table load be interruptible by a M-mode interrupt?

My reading of the spec is that that is allowed, since mepc should be set
to the address for the table load and scause.sinhv should remain set.
When the M-mode handler does an mret leaving M-mode, that will see
scause.sinhv set, and so resume the supervisor table load. Is my
understanding correct?

BTW, I can't find an explicit statement that xepc is set to the table
load address on a trap while xcause.xinhv is set. The closest is the
line "If there is an access exception on the table load, both xtval and
xepc holds the faulting address." I can't see a statement that leads to
that inference. Am I missing something?

pashenden, 2022-04-28:
Related to this, I presume that if the Sdext debug spec is implemented, dpc should be set to the table load address on entry to debug mode while doing the table load, and dret should check the xcause.xinhv bit for the privilege level indicated by dcsr.prv and use dpc either to resume the table load or update the pc and return.

dansmathers, 2022-05-04:
related to issue #228.

kasanovic, 2022-07-19:
There has been text updates since this issue was raised.
In section 5.4, "If there is an access exception on the table fetch, xepc is written with the faulting address. "
This seems quite explicit?

closed 2022-07-19 by kasanovic