riscv/riscv-fast-interrupt #235: is xcause.pil updated on exceptions as well?

dansmathers, 2022-05-04:
I assume if an exception occurs during an interrupt that xcause.pil is updated.  I only see mention in the spec under the section "Changes to xcause CSRs" that xcause.pil is updated due to an interrupt.   Should this be clarified in the spec?

kasanovic, 2022-05-10:
Clarified in #236.

dansmathers, 2022-05-11:
The priv spec uses trap to mean both interrupts and exceptions so the intent of the wording change was to make clear that exceptions will also update xpil.

I would expect xcause.pil= pre-exception "former" mintstatus.xil.  

e.g., for a horizontal (m-mode to m-mode) exception where the hart was running at interrupt level ("former il"), I would expect mcause.pil= former il=current il because horizontal sync exception traps are serviced with the same interrupt level as the instruction that raised the exception.  

The spec says xinhv is set by hw at start of hw vectoring and cleared by hw and end of successful hw vectoring so xinhv I suppose saying that xcause.xinhv is "updated" on traps isn't precise since that field is only set/cleared by hw vectoring logic or csr write to xcause.

kasanovic, 2022-11-08:
Closed with https://github.com/riscv/riscv-fast-interrupt/pull/287

closed 2022-11-08 by kasanovic