riscv/riscv-fast-interrupt #244: Debug trigger behavior

divan0v, 2022-06-09:
Clarify, please, debug trigger behavior for the following case. Let's all clicinttrig regs are implemented as hard-wired zero, i.e. interrupt triggers are not implemented. Let's address of some interrupt (with clicintattr[i].shv=1) is equal to 'A'. Should match control trigger with match type 'execute address' raised when hardware vectoring is occurred for address 'A'?

divan0v, 2022-06-18:
Sorry for inaccurate question. A more clear question is: 
1. Should match control trigger with match type 'execute address' raise when hardware vectoring is occurred for xtvt[i]?
2. Should match control trigger with match type 'load address or load data' raise when hardware vectoring is occurred for xtvt[i]?

kasanovic, 2022-07-19:
1). is correct.  The hardware table read is treated as an instruction fetch.  This is an optional hardware feature whether breakpoints/triggers are supported on hardware table reads.  See #228.

If the table was read with a regular load instruction, then this is treated like any other load wrt debug.

closed 2022-07-19 by kasanovic