riscv/riscv-fast-interrupt #267: Accessing mscratchcsw and mscratchcswl

silabs-oysteink, 2022-09-05:
The specification states the following in chapter 13.2:
`Though designed to be used with csrrw instructions, these CSRs can be accessed with any CSR instruction.`

And a few lines below the following:
`The implementation of the non-CSRRW variants (when either rd or rs1 is x0 or when rs1 is an immediate operand) on xscratchcsw is not defined/reserved.`

How should this be interpreted, are non-CSRRW accesses reserved or how should they behave?

dansmathers, 2022-09-05:
linking  issue #51 
I think we meant that the behavior of non-CSRRW accesses are implementation-defined.  That phrasing seems to be used several times in the privilege spec so I'll create a pull to update the CLIC text.

silabs-oysteink, 2022-09-28:
Thanks for the answer, this made it more clear.

closed 2022-09-28 by silabs-oysteink