riscv/riscv-fast-interrupt #280: Access to mnxti with non-zero immediates

silabs-oysteink, 2022-10-18:
The current spec states:

> Note: Use of xnxti with non-zero uimm values for bits 0, 2, and 4 are reserved for future use.

Since the mnxti CSR may be accessed with all CSR instructions, the write data may also come from a GPR. Should an access to mnxti be declared illegal only if an immediate has any bits of 0/2/4 set, or in general if any of those bits are written to 1?

dansmathers, 2022-10-18:
software doesn't need to set bits 0, 2, 4 because those bits are WPRI in xstatus.  There is a post 1.0 proposal to have a different CLIC behavior based on potentially using one of those bits. see  #101 
Reserving those bits in the spec for future use allows for forward software compatibility.   I don't think the spec is requiring anything from hardware, i.e. hardware doesn't need to add logic to check those bits.  but we can discuss in our next TG meeting.

kasanovic, 2022-10-25:
Saying these values are reserved indicates that software should not use them.  The hardware is not required to have a specific response.   It is legal, but not required, to trap on these values.  Note that trapping on a register value would be the first case where a core would trap on a data value written to a CSR.  These bits may be used in future standards.

We should probably reserve use of CSR instructions other than CSRRSI/CSRRCI with this CSR.

kasanovic, 2022-11-08:
Closed with https://github.com/riscv/riscv-fast-interrupt/pull/283

closed 2022-11-08 by kasanovic