riscv/riscv-fast-interrupt #282: CSR allocation (and tentative address): `mclicbase`

a4lg, 2022-10-25:
Current documentation states that `mclicbase` CSR would have an address of `0x3??` but `0x3??` indicates machine-mode read-write CSRs.  It seems inappropriate for a read-only CSR.

Also, similar general purpose configuration data structure-pointing CSR `mconfigptr` (a part of the privileged architecture version 1.12) has an address of `0xf15`.

If this CSR is going to be preserved, `0xf??` would be better than the current tentative address `0x3??`.

Strictly speaking, `xintstatus` are also read-only but their addresses are okay to me (represents dynamic state information and updated by other CSR updates).

dansmathers, 2022-10-25:
We are planning on moving these to parameters instead of CSRs.  related to closed issue #80.
The mclicbase register and clicinfo are likely to be replaced by the general discovery mechanism that is in development.

But we will leave this issue open until that is done.   your point about addresses is taken.  similar to #88 where intstatus are read-only but currently allocated in the r/w CSR range.

kasanovic, 2022-10-25:
These should be reallocated to appropriate CSR ranges.

a4lg, 2022-10-26:
Hmm, `xintstatus` also need to change (as pointed out in #88) considering CSRs containing dynamic state information like `vl` and `vtype` from `V` extension uses `0xC??`, a `URO` CSR address range.

I confirmed a PR which removes `mclicbase` (which is okay).

kasanovic, 2022-11-08:
Closed with https://github.com/riscv/riscv-fast-interrupt/pull/284 and https://github.com/riscv/riscv-fast-interrupt/pull/285

closed 2022-11-08 by kasanovic