riscv/riscv-fast-interrupt #289: Semantics of reserved CSR accesses

JamesKenneyImperas, 2022-11-15:
Some types of access to `xnxti`, `xscratchcsw` and `xscratchcswl` CSRs are specified to be _reserved_. Can you clarify what _reserved_ means in this context? Should an Illegal Instruction trap be generated?

Thanks.

dansmathers, 2022-11-15:
We were trying to be consistent with the RISC-V unprivileged and privileged specs use of reserved that the reserved behavior is currently undefined and saved for future behavior.  the reserved behavior is unspecified.  e.g., some platforms may generate illegal instruction traps, others may do nothing or whatever.

Here's text from the unprivileged spec for reserved encodings:
_Reserved encodings are currently not defined but are saved for future standard extensions.
The behavior upon decoding a reserved instruction is unspecified.
Some platforms may require that opcodes reserved for standard use raise an illegal-instruction
exception. Other platforms may permit reserved opcode space be used for non-conforming extensions._

Krste thought reserved was well defined, but maybe we should add a note?
linking issue #280 where Krste discusses meaning of reserved:
"Saying these values are reserved indicates that software should not use them. The hardware is not required to have a specific response. It is legal, but not required, to trap on these values. Note that trapping on a register value would be the first case where a core would trap on a data value written to a CSR. These bits may be used in future standards.
We should probably reserve use of CSR instructions other than CSRRSI/CSRRCI with this CSR."

JamesKenneyImperas, 2022-11-16:
Hi Dan,

Ok. I saw from reading the Privileged Specification that in many cases the behavior of reserved features is defined; for example, when talking about `mnstatus`, the Privileged Specification says:

_The other bits in `mnstatus` are reserved; software should write zeros and hardware implementations should return zeros._

So I was thinking that there was perhaps something similar here. As a reference point, in the Arm world they have the notion of _constrained unpredictable_ to cover things like this: for everything equivalent to a RISC-V _reserved_ feature, the implemented behavior _must be one of an explicitly-defined set_. In the case of a reserved access to `xscratchcswl`, they might for example list these possible behaviors:
- the instruction is treated as a `NOP`
- the instruction causes an Illegal Instruction trap
- the instruction behavior follows normal CSR access rules, although these are not useful

A compliant implementation must do one of these things, but application code must not rely on any one of these behaviors. I think this notion is very useful because it puts clearly defined limits on what must occur in such a case, explicitly excluding anything that could (for example) open a security hole or modify state in an unexpected way. Note that it explicitly excludes using the _reserved_ instruction for some non-conforming extension; would this ever be a good idea here?

Thanks,

James.

kasanovic, 2022-11-22:
As part of general spec cleanup after the adoc conversion is complete, we will revisit the definitions in the unprivileged and privileged manuals.   We do want to explicitly not disallow non-conforming in the specs,  as for example, future standard extensions will appear to be non-conforming until they are ratified, though we could introduce a type of behavior to represent this.  We want to preserve modularity in the specs so we do not have to modify a spec after ratification.

For now, for these cases, reserved does not require an illegal instruction exception be raised.

dansmathers, 2022-12-05:
created issue https://github.com/riscv/riscv-isa-manual/issues/924

dansmathers, 2023-01-17:
Discussed again in TG meeting 1/17/23.   reserved is catch-all for don’t do it and if you do it anything can happen.