riscv/riscv-fast-interrupt #303: How are CLIC-only CSRs and fields handled in CLINT mode?

kasanovic, 2023-02-14:
State is preserved on CLIC<->CLINT mode changes.

Text says they read as zero, but no description of behavior on writes.  Should probably not be possible to write a CLIC-only CSR in CLINT mode.

Could alternatively trap on any access (read or write) to CLIC-only CSRs in CLINT mode.

dansmathers, 2023-07-14:
saw this in an email thread about upcoming changes to be included in priv spec 1.13.  "I was under the impression that it will no longer require illegal instruction exceptions on attempts to access undefined CSRs."

If priv spec removes illegal instruction exceptions on attempts to access undefined CSRs, probably shouldn't trap on accesses to CLIC-only CSRs in CLINT mode?

dansmathers, 2023-08-15:
Note this behavior in CLIC mode to CLINT only CSRs: Writes to xie/xip will be ignored and will not trap (i.e., no access faults). xie/xip always appear to be zero in CLIC mode.

kasanovic, 2023-10-10:
When not in CLIC mode, behavior of CLIC-only CSRs should be same as if CLIC was not implemented, so should be treated as a "reserved" operation.   The value in the CSRs after switching from CLIC->CLINT->CLIC mode should be treated as "reserved" also, i.e., the value is undefined.

commit 9a16202d (2023-12-19): Clarify behavior of CSRs when switching between CLIC and CLINT modes
Closes #303
closed 2023-12-19 by kasanovic