riscv/riscv-fast-interrupt #308: Any higher priority interrupt causes xnxti service loop exit

MarkHillHuawei, 2023-03-03:
We have been studying the reference 11.1 C-ABI Trampoline Code and noticed that any pre-emption that occurs during the service_loop for horizontal interrupts will inevitably leave mcause.pil raised to the current interrupt level when it resumes. Then, when the mnxti at step (12) executes, the check that clic.level>mcause.pil will fail even when there are interrupts to service at the current level (or any lower level which is above the value of mcause.pil on entry to the CSR). The service loop will therefore exit and an unnecessary restore and save sequence initiated.

We see a few options:
1) Decide that was "unlucky" and hope the situation does not occur frequently enough to care, not sure that holds in an embedded highly interrupt driven system.

2) Immediately after the C ABI Routine returns at step (11), and prior to the mnxti at step (12), reload the value of mcause saved on entry,  disable interrupts and then restore the value of mcause to the CSR, then proceed to step 12 (which will re-enable interrupts). Disabling the interrupts ensures that no pre-emption can occur between the restoration of mcause and the mnxti. However, this costs two CSRs and a load so will add a few cycles to every iteration of the service loop.

3) Add a previous-previous interrupt level to mcause (ppil?). On ISR entry pil is saved to ppil before it is updated with the interrupt level of the pre-empted code.  On a mret, the new interrupt level is taken from pil, and pil is set to ppil. This ensures that any pre-emption that occurs will restore pil to the value it had prior to entry to the pre-empting ISR.  When pre-emption occurs ppil will be left modified but there is no test against ppil during the service_loop and the value that ppil had on entry to the ISR will be restored by the mcause. This "preserves" the chain of interrupt levels however deep the nesting goes so only one additional field is needed. This solution has the benefit that no additional instructions are needed anywhere in the ISR.

BTW I have also noticed a lack of consistency about whether it is mcause.mpil/xcause.xpil (10 times) or mcause.pil/xcause.pil (13 times). Probably worth tidying up and settling on one or the other. I have no preference.

MarkHillHuawei, 2023-03-03:
Option 4!

Define a behavior for xnxti when the the source operand is a register rather than an immediate. This would always set the enable bit but would check the clic.level against bits 23:16 of the source operand instead of relying on xcause.xpil. The service_loop then passes in the value mcause had on ISR entry.  This approach avoids the need for the two CSRs used in Option 2 to disable interrupts and restore mcause. 

It is also possible to remove the load from the service loop by keeping the mcause in a callee-save register but this does mean an additional register needs to be saved and restored on entry/exit to the ISR so may not be a "win"

MarkHillHuawei, 2023-04-12:
Yes the value of pil on entry to the software loop handler is lost, when pre-emption occurs; I could have been more explicit about that. To clarify, you can't address the issue in the pre-empting handler because it must set pil to the interrupt level of the service loop, not the value of pil just prior to the pre-emption occurring.

kasanovic, 2023-06-06:
We will explore the option of the new CSR form to reduce overhead.  There might be some complications with CSR instruction encoding and semantics, as this should not be viewed as writing a CSR.

kasanovic, 2023-12-19:
Closed with #344

closed 2023-12-19 by kasanovic