riscv/riscv-fast-interrupt #311: behavior/arrangement of mcliccfg.nlbits/scliccfg.nlbits/ucliccfg.nlbits

dansmathers, 2023-03-14:
separating additional question raised in issue #304 into a new issue:
Additionally, can you please clarify the behavior of the xcliccfg.nlbits fields; specifically:

Are xcliccfg.nlbits fields independent of each other, or are they aliases of the same thing? and:
If xcliccfg.nlbits fields are independent, is it the case that mcliccfg.nlbits is used to determine the level of interrupts targeting M-mode, scliccfg.nlbits is used to determine the level of interrupts targeting S-mode and ucliccfg.nlbits is used to determine the level of interrupts targeting U-mode?
Thanks.

kasanovic, 2023-03-28:
Reformatted cliccfg in https://github.com/riscv/riscv-fast-interrupt/commit/9d0784113d5ac87296ab4b2c1844a2e4ef833c8e

dansmathers, 2023-03-28:
closed by #316 and text updates.

closed 2023-03-28 by dansmathers