riscv/riscv-fast-interrupt #317: Clarification on trampoline examples and callee-save expectation for non-x registers

christian-herber-nxp, 2023-03-29:
The spec states that 

> The attribute tells the C compiler to use callee-save for all registers, so the handler has to "pay as it goes" to
> use registers, and only save the full caller-save set if it makes a nested regular C call.

As I understand, the trampoline examples do make regular C calls to the handlers, and save the corresponding `x` registers. They do not save `f` registers. Same argument could be made for `v` registers, although this is more far-fetched.
Are those examples intended for no F-extension systems. Should handlers not use `f` registers? If you don't explicitly disallow it, it will be used by somebody.

This issue is not a concrete suggestion, but first of all a request for comments.

kasanovic, 2023-04-11:
Yes, we should add a comment that the example code doesn't account for presence of f or v registers.

dansmathers, 2023-05-09:
closed by pull #326