riscv/riscv-fast-interrupt #321: Clearing xintthresh when fewer than 8 threshold bits are implemented

JamesKenneyImperas, 2023-04-05:
In various places it is stated that interrupt context returns to a lower privilege level set `xintthresh=0`. However section 9.2.1 (_smclicconfig Changes to Interrupt-Level Threshold (xintthresh) CSRs_) says that some bits of `xintthresh.th` may be hard-wired to 1:

> For example, if CLICINTCTLBITS is 1 and INTTHRESHBITS is 2, interrupts can be set to level 127 or 255 and xintthresh.th can be set to 63, 127, 191, or 255.

Can you please clarify return behavior when fewer than 8 threshold bits are implemented?

Thanks.

kasanovic, 2023-04-11:
The xintthresh should be set to the lowest supported value, not necessarily zero.  The text needs to be updated in several places.

dansmathers, 2023-05-09:
closed by #327