riscv/riscv-fast-interrupt #333: Question about xinhv

hwliu-tommy, 2023-05-25:
Hi, 

I'm curious about why inhv is defined as a working state at *x*cause.*x*inhv (*x*=previous privilege)?

> When these interrupts are taken, the hart switches to the handler’s privilege mode, and besides the trap side effects described in this and the privileged specification (e.g. update xintstatus, xcause, xstatus fields including clearing xstatus.xie), also sets the hardware vectoring bit xinhv in xcause of the handler privilege mode.  (section 7.2.1)

> If the fetch is successful, the hart clears the low bit of the handler address, sets the PC to this handler address, then clears the xinhv bit in xcause of the handler privilege mode. (section 7.2.1)

> When returning from an xret instruction, the xinhv bit modifies behavior as follows:
If the xinhv bit of the previous privilege mode is set, the hart resumes the trap handler memory access to retrieve the function pointer for vectoring with permissions corresponding to the previous privilege mode. The trap handler function address is obtained from the current privilege mode’s xepc with the low bits of the address cleared to force the access to be naturally aligned to an XLEN/8-byte table entry. If the fetch is successful, the hart clears the low bit of the handler address, sets the PC to this handler address, then clears the xinhv bit in xcause of the handler privilege mode. (section 7.2.1)

For the case that an exception (handled in Machine) happens during hardware vectoring of Supervisor interrupt.
In this case, M handler knows the trap is happened during hardware vectoring by checking scause.sinhv, and MRET honors scause.sinhv to restart the hardware vectoring. 
But as far as I know, normal machine handler only checks the machine trap CSRs, mcause, mepc and mtval, it might have to further access scause for check inhv and the other fields in scause are meaningless for it.
Though sinhv is in scause, this field is mostly used by machine handler, supervisor handler would never observe scause.sinhv = 1 except nested trap cases.  
And if sinhv is writable by S-mode, would it be a risk that if a M-mode trap happens right after a S-mode instruction set scause.sinhv?

If the inhv state is implemented as a sampled state, hardware sets it on trap taken with mcause, mepc update .
(Keep the inhv state in handling privilege instead of previous privilege.  Update when machine trap taken, if trap happens during hardware vectoring, then set minhv, otherwise, clear minhv, and MRET honors minhv)
Then the horizontal trap can be handled by software by backup minhv and recover it after returned from the preempted trap handler.
If the minhv is in mcause or mepc, it can push/pop with mcause.exccode, it won't increase the code of critical section.

Thanks,
Tommy

dansmathers, 2023-05-25:
Hi,
An example of a functional inhv use case is with demand paging.  an s-mode hw-vectored interrupt attempts to do a table fetch (scause.inhv is set by the hw at the beginning of the fetch) but the table is not paged in so a m-mode page fault exception occurs.  Note that the m-mode handler does not need to check scause.inhv, it only needs to determine the bad virtual address to page in and finish with an mret.  mepc is pointing to the table entry, and since mret honors the previous priv xcause (scause.inhv) the table entry fetch (again with s-mode priv) will be treated as an address instead of as an instruction.  

> And if sinhv is writable by S-mode, would it be a risk that if a M-mode trap happens right after a S-mode instruction set scause.sinhv?
I think the use model is s-mode software would set sepc to point to a table entry containing addresses before setting scause.sinhv if s-mode software ever had a valid reason to do that.  for security risks, I think this is similar to concerns about JVT in the Zc extension  https://github.com/riscv/riscv-code-size-reduction/issues/123

I'll bring this up to discuss in the next fast-interrupt TG meeting to make sure I didn't miss anything but feel free to add more comments if you like.  Thanks.

hwliu-tommy, 2023-05-26:
Thank you for replying.

I understand the use cases of inhv, and agree that inhv are not required for the current design works fine in these case.
But I'm still curious about why we decided to save inhv as working state in scause.sinhv instead of a sampled state in mcause.minhv (sampled on trap taken), which is more similar to the original trap model. (just my personal view)

Here's my concept about using mcause.minhv.

> If the inhv state is implemented as a sampled state, hardware sets it on trap taken with mcause, mepc update .
(Keep the inhv state in handling privilege instead of previous privilege. Update when machine trap taken, if trap happens during hardware vectoring, then set minhv, otherwise, clear minhv, and MRET honors minhv)
Then the horizontal trap can be handled by software by backup minhv and recover it after returned from the preempted trap handler.
If the minhv is in mcause or mepc, it can push/pop with mcause.exccode, it won't increase the code of critical section.
The minhv is treated as an decorator of mepc, so they are always modified together.

My understandings are:

- with scause.sinhv, M handler only has to guarantee the pc and inhv when the next time it returns to S mode, so horizontal traps can ignore just inhv. But if it wants to switch to other S mode program (this might happen on unrecoverable fault or maybe context switch), it has to modify/backup scause.sinhv. And the handler has to guarantee MRET sees the correct sinhv/mepc pair, this is risky because they are managed separately.
- with mcause.minhv, M handler has to backup minhv before any horizontal trap, but this can be merged with mcause backup.  Since mcause/mepc are mostly push/pop together, it can reduce the risk of managing minhv/mepc pair incorrectly. 

Both methods seems to work fine in the common case (resume vectoring).
Maybe I missed something, but mcause.minhv looks better for me because it can be simply managed with mcause/mepc.
So I'm curious about why we decided to implement scause.sinhv.

Thanks,
Tommy

dansmathers, 2023-05-27:
Hi,
Suppose a u-mode interrupt starts hardware vector table lookup and it is interrupted by a s-mode interrupt that starts the hardware vector table lookup and the s-mode page has a page fault into m-mode.   since tables can be located in memory space, e.g. dram, the u-mode hardware vector table lookup might take a while so this isn't too difficult of a situation to create.   With your recommendation, mcause.inhv and prevpriv would show s-mode vector table lookup had the fault, but u-mode vector table lookup sampled state is lost/has to be stored elsewhere.   In the current system, the xpriv sampled state is stored in xcause, so for the above example, ucause.inhv is set, scause.inhv is set.   and after the m-mode exception is handled, mret references scause.inhv to resume the s-mode hardware vector table lookup and at the end of the s-mode interrupt, sret references ucause.inhv to resume the u-mode hardware table lookup.  
Maybe you can comment how your method would handle this case.

We can discuss in our next TG meeting your understandings.

hwliu-tommy, 2023-05-29:
Many thanks for the comments.

> I think the problem with your suggestion is that exception handlers then have to check both inhv and mcause.mpp to know if it was vertical or horizontal (recoverable or not). 
The current location of inhv in the spec means if inhv is set, it was horizontal and unrecoverable. 

I missed this case, thank you for pointing out. 
The spec's behavior (xinhv, x=faulting priv) helps for recognizing unrecoverable traps, but it is unachievable with the proposed behavior (xinhv, x=handling priv). 

> I'm not sure I see your point of the problems m handler having to worry about scause.inhv. It seems like unrecoverable faults or context switches are going to require a lot of extra software anyway so not sure it makes sense to optimize those cases.

Yes, these are my concerns.
And another consideration is that the proposed version (xinhv, x=handling privilege) works more like the original model, hardware only updates M-mode trap CSRs when taking the trap that are handled by M-mode.

Thank you for answering these questions.

Tommy

hwliu-tommy, 2023-05-31:
> I missed this case, thank you for pointing out. The spec's behavior (xinhv, x=faulting priv) helps for recognizing unrecoverable traps, but it is unachievable with the proposed behavior (xinhv, x=handling priv).

I just came up with a case that for breakpoint to debug happened during hareware vectoring, using xinhv (x=faulting priv) may save an extra xinhv bit for debug mode (comparing to xinhv, x=handling priv).

kasanovic, 2023-06-06:
xinhv is in the privilege mode of the handler, i.e., same as epc.

Need to find any contradictory text and fix.

dansmathers, 2023-06-20:
my reading of the spec is xinhv is the privilege mode of the interrupt handler, i.e., same as interrupt handler epc.  but if an exception handler occurs due to a page table fault while the interrupt handler is accessing the vector table in a higher privilege mode, is inhv set in the privilege mode of the exception handler or the interrupt handler?  Let's discuss in our next meeting.

dansmathers, 2023-07-14:
adding comments from TG 6/20/23 - summary xinhv only useful to the exception handler and interrupt handler shouldn’t really see it.   should probably change where invh is set but need to work thru scenarios.

here are the notes from the discussion:
Example: Suppose in u-mode.  get s-mode interrupt and try to read table.  table read causes m-mode exception. so have both s-mode interrupt handler and m-mode exception handler.  currently sinhv is set in interrupt handler.  the mode of the handler is what text says.  the thing that handles the table read exception, just sees a table read exception.  it will see sinhv is set so reason for exception was a table read in s-mode.   when ret from m-mode exception handler, normally don’t touch sinhv.  what may be wrong is exception will create mepc which points at the table entry.   

Some times page tables.  s-mode running, set to s-mode handler handles page fault.  at end, just does a ret.  we should walk thru the scenario of a page handler.  idea is paging a table.  even if m-mode does nothing else, want it to report it.  but to resume assumes s-mode/vs-mode to bring table back in and resume.

closed 2023-08-01 by kasanovic

eroom1966, 2023-09-06:
I would like to repeat a question (asked previously) regarding the clearing of the inhv due to an xRET
Previously @kasanovic said here

https://github.com/riscv/riscv-fast-interrupt/issues/154#issuecomment-839825594
_Clearing inhv: "If the fetch is successful, the processor clears the low bit of the handler address, sets the PC to this handler address, **then clears the xinhv bit in xcause**."_

This seems to have been removed from the specification by @dansmathers here
https://github.com/riscv/riscv-fast-interrupt/pull/341
https://github.com/riscv/riscv-fast-interrupt/pull/341/commits/d9ecea466ae5c7e16cd6e31b2ac432f682fd12f2

So my question reverts back to the original question posed 
https://github.com/riscv/riscv-fast-interrupt/issues/154#issuecomment-839507063
and answered in here to indicate that it is cleared
https://github.com/riscv/riscv-fast-interrupt/issues/154#issuecomment-839825594

has the behavior changed ?
if not and it is cleared, where do I find that in the updated specification - it is not clear to me.

thx