riscv/riscv-fast-interrupt #348: xcause.inhv conflicts with enabling CLIC per mode

JamesKenneyImperas, 2023-08-18:
A recent change to the specification redefines the `inhv` field so that it is held with the modeX _exception_ mode instead of the modeY _interrupt_ mode. This makes sense because this information is required by the exception handler, not the interrupt handler.

Currently the CLIC can only be globally enabled, but the specification states that in future this might change to allow CLIC to be enabled per privilege level. The new definition will therefore impose an extra constraint that if CLIC is to be enabled at any privilege level, it must also be enabled at all higher privilege levels (because the `inhv` bit only exists for a privilege level if the CLIC is enabled for that privilege level).

It feels to me that with this change the `inhv` bit is in the wrong place: it should really be in a fixed position in `mstatus`/`mstatush` instead.

JamesKenneyImperas, 2023-10-11:
Hi Krste,

Okay, but some sections are going to need some careful rework later. For example, section 7.2.4 says this:

> For backwards compatibility in systems supporting both CLINT and CLIC modes, when switching to CLINT mode the new CLIC xcause state fields (xinhv and xpil) are zeroed.

This doesn't work when CLIC is enabled per privilege mode, because the `xinhv` field is potentially owned by a different privilege mode.

> In CLIC mode, when a trap is taken, xcause has the CLIC format and the xcause fields are updated. On the other hand, when not in CLIC mode, xcause has the CLINT mode format.

This clearly won't be true when the CLIC is enabled per privilege mode if the higher privilege level that "owns" `xinhv` is in CLINT mode.

Thanks.

christian-herber-nxp, 2025-05-07:
inhv removed https://github.com/riscv/riscv-fast-interrupt/pull/513

closed 2025-05-07 by christian-herber-nxp