riscv/riscv-fast-interrupt #355: Is there a specification for mnstatus.mpp and mnstatus.mpie?

JamesKenneyImperas, 2023-09-01:
Section 4 says:

> The `mncause.mpp` and `mncause.mpie` fields mirror the `mnstatus.mpp` and `mnstatus.mpie` fields, and are aliased into `mncause`.

The latest RNMI specification I've found (0.4) states that `mnstatus` has fields `MNPP`, `MNPV` and `NMIE` only. Is there a more recent version of the RNMI specification describing the fields mentioned here? If so, can you point me at it?

Thanks.

dansmathers, 2023-09-11:
I only found this.
https://github.com/riscv/riscv-isa-manual/blob/main/src/rnmi.adoc
Plan to discuss this question in TG meeting.

kasanovic, 2023-09-26:
RNMI spec has been stalled waiting ACT and SAIL resources.  We are trying to unblock the RNMI spec now.

kasanovic, 2023-09-26:
There should actually be no interaction between resumable NMI (RNMI) and the fast interrupts section, and the sections regarding NMI (both non-resumable and resumable) will be re-written.  The RNMI CSRs are separate from the fast interrupts CSRs and so the state does not need to be saved in the RNMI registers.  This makes the design of the RNMI extension orthogonal to the core's general interrupt scheme.

closed 2023-10-10 by kasanovic