riscv/riscv-fast-interrupt #360: Smstateen bit

JamesKenneyImperas, 2023-10-03:
This specification does not currently say _which_ `Smstateen` bit is used to control access to S-mode CSRs - this needs to be defined.

Thanks.

kasanovic, 2023-10-10:
Yes, we'll be requesting a tentative assignment from AR.

dansmathers, 2023-10-11:
reply from AR:  This was discussed some in this week's AR committee meeting.  At a minimum it looks like you _will_ need a bit in mstateen, and tbd whether also in sstateen.  John Hauser will review this email thread and let all of us know what he believes is necessary (and what would be the appropriate bit assignment).  Feel free to ping him if you haven't heard back after a little while.

dansmathers, 2024-02-10:
My apologies, Dan.  Your original request back in October didn't make
it onto my radar at the time.

I haven't had time to consider the question of whether CLIC should
really have a stateen bit, although superficially it seems likely.  But
it's trivial to go ahead and assign a tentative bit number for CLIC,
independent of resolving whether CLIC actually needs it and will keep
it in the end.

If there are no objections, be it therefore resolved that CLIC is
tentatively allocated bit 53 in the stateen0 registers, with the
presumed bit name of "CLIC".

dansmathers, 2024-02-10:
spec updated 2/9/24

closed 2024-02-10 by dansmathers