riscv/riscv-fast-interrupt #361: CLICINTCTLBITS significance

wizflame, 2023-10-06:
If the following scenarios takes place
CLICINTCTLBITS = 3
INTTHRESHBITS = 4
THRESHOLD = {4'h0, 4'hf}                     only first 4 bits are writable
CLICINTCTL(intr_id=i) = {3'h0, 5'h1f}    only first 3 bits are writable
CLICINTIE(intr_id=i) = 8'h1;
assume that mode is same as current privilege and interrupt is level type. Then when the external interrupt occurs what is the expected behaviour?
Since the level of the interrupt is set at 0 for the writable field but 5 bits from LSB are read only 1. Then in this scenario should the interrupt occur with level as 8'h1f or should it not occur at all?

dansmathers, 2023-10-06:
in section 4.5:
>Next, the smclicconfig extension defines how to split the clicintctl value into interrupt level and interrupt priority
For your example I will assume all clicintctl bits represent interrupt level.  just want to point out that there is an option for all interrupts to be at level 255 and clicintctl bits just represent priority.

in section 4.9.1:
> Current  |      CLIC                 |->      Current'          Previous
> p/ie/il    | priv level   id         |->    p/ie/il  pc  cde   pp/il/ie epc
> P  1  L    | nP=P   L<nL    id  |->    P 0  nL  V   id      P  L  1  pc  # Horizontal interrupt taken

so if I decode your scenario correctly: the interrupt is pending, it is enabled, the level is 8'h1f, next priv == current priv, if nL (8'h1f) > L (max (intstatus.xil,xintthresh=8'hf)), a horizontal interrupt is taken.
 
so in your scenario the interrupt should occur with level 8'h1f if 8'h1f is greater than the current interrupt level.

wizflame, 2023-10-09:
Great Thanks!

closed 2023-10-09 by wizflame