riscv/riscv-fast-interrupt #371: clicinttrig behavior when the interrupt is claimed by xnxti

hwliu-tommy, 2024-02-16:
Hi, 

The description of clicinttrig says that the trigger is signaled (for triggering breakpoint) when the selected interrupt is taken. 
> A trigger is signaled to the debug module if an interrupt is taken and the interrupt code matches a
> clicinttrig[i].interrupt_number and the associated clicinttrig[i].enable is set.

If the interrupt (selected by clicinttrig) is claimed by xnxti, should the trigger also be signaled?

Thanks

dansmathers, 2024-02-20:
Good question.  
I can imagine a scenario where a user may want to send a signal to TM external trigger (to increment custom event counters or outputs for use by external logic) every time a selected interrupt is taken.  Looking at the C-ABI trampoline Code, however, it would be hard to get an accurate count.  
If xnxti didn't signal a trigger, only the initial interrupt at irq_enter would be signaled and interrupts taken using xnxti would be missed.  
If xnxti did signal a trigger, there might be double counting if the initial interrupt was also the highest current interrupt at label 6.  
Seems like we might want to add another bit to clicinttrig to control that behavior and add text to clarify the behavior.

hwliu-tommy, 2024-02-27:
The update resolves my question.
Thank you!

dansmathers, 2024-03-02:
reopening until pull #372 gets into spec for tracking purposes.  (planning on discussing and approving in next TG meeting march 4).

dansmathers, 2024-03-04:
closed by #372

closed 2024-03-04 by dansmathers