riscv/riscv-fast-interrupt #376: Trampoline for Preemptible Inline Handlers

christian-herber-nxp, 2024-03-05:
In the section "Trampoline for Preemptible Inline Handlers":

as part of irq_enter, the following check is performed.
csrrsi a0, mnxti, MIE        # Get highest interrupt, enable interrupts.
beqz a0, exit

Two questions:
1. Is it really beneficial to check for a higher prio interrupt at this point? It doesn't really reduce a worst-case latency by a lot, e.g. if the interrupt arrives two cycles later, it still has to sit through the entire ISR. On the other hand, you are adding two instructions into the path of every interrupt.
2. Checking for 0. I understand this could happen if a level based interrupt is retracted. But this would be a rare case, and accepting a short handler in this case would be better than having this branch instruction always.

dansmathers, 2024-03-05:
This is the preemptible example so it does need to enable mstatus.mie to allow higher level interrupts to preempt. I think the beqz is then really just error checking if the interrupt was retracted or an shv interrupt showed up.   
So I think we could replace 
```
csrrsi a0, mnxti, MIE # Get highest interrupt, enable interrupts.
beqz a0, exit
```
with
`csrrsi x0, mstatus, MIE`
if we wanted to avoid the extra beqz.

But it is just normative text so isn't too important.  Would you like me to add a comment that we could avoid the beqz?

dansmathers, 2024-03-07:
sorry. yes, non-normative.  I corrected my comment above.  good point about moving to an appendix

dansmathers, 2024-03-07:
I see both AIA and priv spec use inline commentary in italics to specify non-normative text.
e.g.
_Commentary on our design decisions, implementation options, and application is formatted as
in this paragraph, and can be skipped if the reader is only interested in the specification itself._

No spec change is required, just noting that chapter 11 (interrupt handling software through Chapter 17 CLIC interrupt ID ordering recommendations) all should be in an appendix/inline commentary.  I'll add a label of spec formatting to this issue that can be resolved during ratification phase but not gate freeze.  Do you agree?

christian-herber-nxp, 2024-03-08:
totally fine

christian-herber-nxp, 2025-05-06:
the sections in question were removed

closed 2025-05-06 by christian-herber-nxp