riscv/riscv-fast-interrupt #385: Allow fixed configuration for some fields

christian-herber-nxp, 2024-03-05:
clicintattr.trig has the following note:

> Some implementations may want to save these bits so only certain trigger types are supported. In this case, these bits become hard-wired to fixed values (WARL).

To me it is not clear if similar behavior is allowed for other config bits. I think we need to allow this. In particular, I think it should be allowed to fix ie, clicintattr.ie, and clicintattr.mode to certain values.

dansmathers, 2024-03-05:
from the priv spec:

> Some read/write CSR fields are only defined for a subset of bit encodings, but allow any value to be
> written while guaranteeing to return a legal value whenever read. Assuming that writing the CSR
> has no other side effects, the range of supported values can be determined by attempting to write
> a desired setting then reading to see if the value was retained. These fields are labeled WARL in
> the register descriptions.

My understanding is that any register defined as WARL can be allowed to be hard-wired to fixed values.  The clic spec currently lists clicintctl and clicintattr as 8-bit WARL fields.  

There is also this statement in the CLIC spec for when interrupts aren't implemented:
> If an input i is not present in the hardware, the corresponding clicintip[i], clicintie[i], clicintattr[i], clicintctl[i] locations appear hardwired to zero.

Is that sufficient or are you asking if we should have the CLIC spec to allow clicintie for some interrupts to be hardwired to 1, i.e., add a statement that clicintie field is also WARL.

dansmathers, 2024-03-07:
@christian-herber-nxp, no problem adding a note for the other WARL fields in clicintctl/clicintattr.  You mentioned clicintattr.ie earlier.  that's not a clicintattr field.  did you mean clicintie which is not currently defined as WARL?

christian-herber-nxp, 2024-03-07:
yes, in that case I think it should be WARL.
In case an interrupt is not implemented, I recall there was a text that it would be read-only zero.

dansmathers, 2024-03-07:
closed by pull #390.

closed 2024-03-07 by dansmathers