riscv/riscv-fast-interrupt #398: Questions on Release 0.9

hirooih, 2024-07-15:
The following are what are unclear from reading the specifications. I hope this helps.

***

Chapter 7. suclic U-mode CLIC extension

Is N-extension still alive?  From RISC-V Spec:

> Preface to Version 20211203
> ...
> Removed the N extension.

It was my understanding that the N extension was removed from the candidate for the extended specification.

***

9.3. CLICINTCTL Parameters

> Only 0 or 8 level bits are currently supported, with other values currently reserved.

0 level bit cannot support the horizontal interrupt preemption which is the most important feature of CLIC. I doubt it is worth supporting.

8 level bits cannot support the interrupt priority.  I guess this might be a patent issue.  When will we be able to support other settings?

> ... `xCLICINTCTLVALUES` would be the set [0-255].

"[0-255]" should be "[1-255]" according to the table on the next page.

> ... when `clicintattr[i].mode` bits 7:4 are all 1s otherwise s-mode.

According to "5.4. CLIC Interrupt Attribute (clicintattr)",  `clicintattr[i].mode` is assigned to bits "7:6" and "11" means Machine mode.  I don't understand what this paragraph means.

> with the lower unimplemented bits treated as hardwired to 1.

The lower unimplemented bits of `clicintctl[i]` should be read as 1s by a CSR instruction?

From the last paragraph:

>  ... For example, when nmbits is 0, ... to clicintctl[i] registers.

I think this sentence should be moved to "10.1.1. CLIC Configuration (xcliccfg registers)".

***

9.4. Additional CLIC Parameters

> CLICLEVELS     2-256

"2-256" should be "1-256" if the "0 level bit" is supported.

> CLICMAXID      12-4095

From where the number "12" comes?  The max ID is 2 in the example on 17.4.  Only one interrupt does not require CLIC features. "1" may be the minimum number.

> INTTHRESHBITS  1-8

All other parameters are prefixed by "CLIC".  Is it intentional that only this parameter is not prefixed by "CLIC"?

***

10.1.2. Specifying Interrupt Privilege Mode

I have no idea why `cliccfg.nmbits` exists.
For example In the following case (and M/U-mode systems without the suclic extension) we can hardware the `clicintattr[i].mode` to "11".

> For example, in M-mode-only systems, only M-mode exists so we do not need any extra bit to represent the supported privilege-modes. In this case, no physically implemented bits are needed in the `clicintattr.mode` and thus `cliccfg.nmbits` is 0 (i.e., `cliccfg.nmbits` can be hardwired to 0).

***

The attached includes some other typo things.

[clic.patch](https://github.com/user-attachments/files/16232715/clic.patch)

hirooih, 2025-01-05:
I've found that some of questions reported in this issue are already fixed.  And I found some answers by myself.

I welcome anyone could to comment on the questions that still open.

***

> Chapter 7. suclic U-mode CLIC extension
>
> Is N-extension still alive? ...

- closed: #403 removed the descriptions about N-extension.

***

> 6.3 (was 9.3). CLICINTCTL Parameters
>
>> Only 0 or 8 level bits are currently supported, with other values currently reserved.

- closed: #403 removed this sentence, too.

***

>> ... xCLICINTCTLVALUES would be the set [0-255].
>
> "[0-255]" should be "[1-255]" according to the table on the next page.

- closed: I was wrong. Although the level 0 behaves as a locally disabled interrupt, the level 0 is still a legal write value for `clicintctl[i]` (only when `xNLBITS` == 8 or (`xCLICINTCTLVALUES` == 8 and `xcliccfg.xnlbits` == 8)).

***

>> ... when clicintattr[i].mode bits 7:4 are all 1s otherwise s-mode.
>
> According to "5.4. CLIC Interrupt Attribute (clicintattr)", `clicintattr[i].mode` is assigned to bits "7:6" and "11" means Machine mode. I don't understand what this paragraph means.

- still open: `clicintattr[i].mode` is assigned to bits "7:6". It has only 2-bits.  "clicintattr[i].mode bits 7:4" must be wrong.
Replace `clicintattr[i].mode`s in this paragraph with `clicintctl[i]`s, and this paragraph makes sense.

***

>> with the lower unimplemented bits treated as hardwired to 1.
>
> The lower unimplemented bits of `clicintctl[i]` should be read as 1s by a CSR instruction?

- still open: My current interpretation is that the answer should be “No”.
The implemented bits of `clicintctl[i]` are the concatenation of level and priority.
The lower unimplemented bits of `clicintctl[i]` are neither the lower bits of level nor the lower bits of priority.
That is, they can be read as 0.

Am I correct?

***

>> ... For example, when `nmbits` is 0, ... to `clicintctl[i]` registers.
>
> I think this sentence should be moved to "10.1.1. CLIC Configuration (`xcliccfg` registers)".

- still open: Or add a reference to 10.1.1.

***

>> CLICLEVELS 2-256
>
> "2-256" should be "1-256" if the "0 level bit" is supported.

- closed: I was wrong.  Only level 0 and 255 is minimum. "2-256" is correct.

***

>> CLICMAXID 12-4095
>
[... 78 lines cut ...]
-For example by limiting MCLICINTCTLVALUES to the set of [240-255] and SCLICINTCTLVALUES to the set of [0-239]
-`clicintattr[__i__].mode` reflects an M-mode setting when `clicintattr[__i__].mode` bits 7:4 are all 1s otherwise S-mode.
+be able to implement fewer `clicintctl[__i__]`.
+For example by limiting MCLICINTCTLVALUES to the set of [240-255] ([0xf0-0xff]) and SCLICINTCTLVALUES to the set of [0-239] ([0x0-0xef])
+`clicintctl[__i__]` reflects an M-mode setting when `clicintctl[__i__]` bits 7:4 are all 1s otherwise S-mode.
 
 The smclicconfig extension uses the following scheme for implementations that choose to implement fewer
 than 8-bits in the `clicintctl` registers is as follows.
```

See above.

```diff
@@ -1949,7 +1948,7 @@ supporting `clicintattr[__i__].mode` control, to instead add additional control
 ----
 Name           Value Range  Description
 CLICANDBASIC   0-1          Implements both CLINT modes too
-CLICPRIVMODES  1-3          Number privilege modes: 1=M,2=M/U,3=M/S/U
+CLICPRIVMODES  1-3          Number of privilege modes: 1=M,2=M/U,3=M/S/U
 CLICLEVELS     2-256        Number of interrupt levels including 0
 CLICMAXID      12-4095      Largest interrupt ID
 
```

Missing "of".

christian-herber-nxp, 2025-05-06:
I ran through the list of issues and i believe all issues are either resolved or no longer relevant with the recent changes.
If I am mistaken, please open dedicated issue for each item in favor of this compound issue.

closed 2025-05-06 by christian-herber-nxp