riscv/riscv-fast-interrupt #400: Make shv a per level property

christian-herber-nxp, 2024-08-15:
This is a proposal to make the following change:

Currently, each interrupt can configured to use the hardware vectoring or not. This does create some challenges, as HW-vectored interrupts do not mix well with the nxti feature.

The change would make the shv bit a per level feature. This would reduce flip flops in implementation, and at the same time prevent HW vectored interrupts from ending the trampoline of the SW-vectored interrupts.

Same behavior could be achieved with the current definition through configuration, but then implementations would be wasting gates.

tovine, 2024-08-22:
This seems reasonable.
Out of curiousity: what were the original reasons to introduce selective hardware vectoring, and how would these be impacted by this change?
My guess would be something like having the ability to use hardware vectoring for time critical interrupts, while having a simpler interrupt scheme for lower priority interrupts such as OS scheduling functions. And for this use case it shouldn't be a big problem - worst case you might have to add another level bit.

christian-herber-nxp, 2025-04-07:
nxti was removed, no longer need this change

closed 2025-04-07 by christian-herber-nxp