riscv/riscv-fast-interrupt #402: Clarify behaviour of xnxti register version in smclicshv

jakubahh, 2024-08-27:
The section "smclicshv Changes to Next Interrupt Handler Address and Interrupt-Enable
CSRs (xnxti)" modifies the behaviour of immediate version of xnxti, but doesn't clearly state how register version should behave

Now there are multiple possible interpretations of the xnxti register version's behaviour.
1, The register version behaves as defined in " Next Interrupt Handler Address and Interrupt-Enable CSRs (xnxti)"
2, The register version behaves the same as immediate version
3, The register version contains the original difference in `clic.level` comparison (clic.level > rs1[23:16]) while returning zero when the incoming interrupt is hardware vectored.

The option 3, seems to be the intended one.
Can the difference between register and immediate version get clear definition in smclicshv?

jb-brelot-nxp, 2024-10-07:
maybe related to https://github.com/riscv/riscv-fast-interrupt/issues/415

christian-herber-nxp, 2025-04-07:
nxti was removed

closed 2025-04-07 by christian-herber-nxp