riscv/riscv-fast-interrupt #405: Conflicting statements about switching from CLIC to CLINT

Timmmm, 2024-09-09:
When you switch from CLIC to CLINT mode the spec says:

>  The other new CLIC xcause fields, xpp and xpie, appear as zero in the xcause CSR but the corresponding state bits in the mstatus register are not cleared.

This doesn't make any sense because it also says:

> The mcause.mpp and mcause.mpie fields typically mirror the mstatus.mpp and mstatus.mpie fields, and are aliased into mcause to reduce context save/restore code.

If they are aliased how do they have different values in `xcause` and `xstatus`?

Also `typically` is very imprecise.

christian-herber-nxp, 2024-09-09:
Totally agree that the typical thing is to be fixed. The behavior in the first quote is the intended one.

jb-brelot-nxp, 2024-09-09:
During the last clic meeting we discussed about this point. Thanks for opening this issue, We need to address it to make it clearer

closed 2024-09-17 by jb-brelot-nxp