riscv/riscv-fast-interrupt #407: The text definition of `xscratchcswl` is incomplete

Timmmm, 2024-09-11:
Without looking at the pseudocode, the definition of `xscratchcswl` is

> This new CSR operates similarly to xscratchcsw except that the swap condition is true when the interrupter and interruptee are not both application tasks or not both interrupt handlers.

However the spec never defines what an application task is. From looking at the pseudocode we can see it is one where the interrupt level is 0, but the text should say that. Something like:

> This new CSR operates similarly to xscratchcsw except that the swap condition is true when interrupt level for the interrupter and interruptee are not both zero and not both non-zero.

jb-brelot-nxp, 2024-09-12:
Hi @Timmmm 

Your change sounds reasonnable. We will disccuss this point on our next Fast Interrupt Task Group, monday the 16th of Sept, 6pm CET.

It would be nice to have you in the meeting.

Cheers

christian-herber-nxp, 2025-04-07:
feature was removed

closed 2025-04-07 by christian-herber-nxp