riscv/riscv-fast-interrupt #411: smclicshv psuedo-code assumes hart has IALIGN=16 and needs to be better documented

james-ball-qualcomm, 2024-09-11:
In "smclicshv Changes to xtvec CSR Mode for CLIC" section, there is pseudo-code showing the least-significant bit of the vector table entry is forced to zero when the hart fetches the entry from the table (the "&~1" in the pseudo-code).
[image]

I have two related issues related to this:

1. I presume this should show "&~1" in implementations that have IALIGN=16 and "&~3" in implementations that have IALIGN=32.
2. We should document outside the pseudo-code that the 1 or 2 least-significant bits of the table entry are forced to zero (depending on IALIGN). We should document those bits as WPRI which means that they are reserved and software must write them with zero.

james-ball-qualcomm, 2024-09-13:
The existing pseudo-code already shows forcing the LSB to 0 (with the implicit assumption of IALIGN=16). If you don't force the 1 LSB (IALIGN=16) or 2 LSBs (IALIGN=32) to zero then what happens if a programmer puts non-zero values there? Presumably it would cause a misaligned instruction address exception since it is a hardware state machine and not a jump instruction.

So, I recommend that the pseudo-code and the text both make it clear that implementations must ignore the bottom 1 or 2 LSBs (depending on IALIGN).

james-ball-qualcomm, 2024-09-13:
Maybe we can talk about this at the next Fast Interrupt TG meeting since I'm not sure I fully understand your comments and perspective?

james-ball-qualcomm, 2024-09-25:
Can we close this now that the related PR has been accepted and merged into the main branch?

jb-brelot-nxp, 2024-09-25:
PR done and merged. can be close