riscv/riscv-fast-interrupt #414: Move register layout definitions into wavedrom

christian-herber-nxp, 2024-09-17:
The privileged spec defines the CSR layouts using wavedrom. The CLIC spec doesn't. This should be aligned.
The layout diagrams do not include descriptions, while the CLIC tables do.

Proposed Resolution:
Move CLIC CSR layouts to wavedrom. Merge the description text with from the layout table with the rest. There might be redundancies, in which the text that doesn't come from the table should take presedence.

wmat, 2024-09-23:
Can you point me at a CSR diagram from the Privileged spec that you would expect the CSRs here to look like?

wmat, 2024-09-23:
Maybe mnstatus from the rnmi chapter 8.3?

christian-herber-nxp, 2024-09-23:
I think we are open to whatever is the preferred presentation by the documentation team. I was looking at the privileged spec itself, i.e. definition of machine mode.

wmat, 2024-10-01:
Can you let me know if the content of this diagram is correct?  I've left the textual description above the diagram for reference:
[image]

christian-herber-nxp, 2024-10-03:
> Can you let me know if the content of this diagram is correct? I've left the textual description above the diagram for reference: ![Screenshot from 2024-10-01 15-54-46](https://private-user-images.githubusercontent.com/57188/372630972-7db4bcda-bcbf-4ecc-abbf-a3f92d40bcd7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc5NTQxNDAsIm5iZiI6MTcyNzk1Mzg0MCwicGF0aCI6Ii81NzE4OC8zNzI2MzA5NzItN2RiNGJjZGEtYmNiZi00ZWNjLWFiYmYtYTNmOTJkNDBiY2Q3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDAzVDExMTA0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRiN2I0YWUxNDY2NGJjNWM4NDMwYWY4ZjI3N2VhYWI5N2Y1YmZlOGFlOGM2YzEwNGVhODU3ZTFiMDdkYzYyMTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.McnhgZvFBzU2-p6zJwYpDt_WEjV_qU2OXaB3PE4eTMI)

looks fine

jb-brelot-nxp, 2024-10-07:
looks ok from my side