riscv/riscv-fast-interrupt #415: Write side-effects for csrrs rd, mnxti, rs1 when rs1 is not x0

MarkHillCodasip, 2024-09-18:
The spec says it follows the usual CSR conventions for write side-effects when accessing xnxti:

https://github.com/riscv/riscv-fast-interrupt/blob/12f72cfbd92dd973f4696bd03d919a98db0c9ab3/src/clic.adoc?plain=1#L979

The "usual CSR conventions" are  defined in the ISA Volume 1 Spec which says that write-side effects occur whenever rs1 is not x0 even if the register contents is zero:

"Note that if rs1 specifies a register other than x0, and that register holds a zero value, the instruction will not action any
attendant per-field side effects, but will action any side effects caused by writing to the entire CSR."

In particular, when following the usual CSR conventions the occurrence of a write-side effect can be determined purely by decoding the opcode and no read of the source operands is needed to detect this.

However this seems to be contradicted by the pseudo code that includes a check for rs1[4:0] being non-zero:

https://github.com/riscv/riscv-fast-interrupt/blob/12f72cfbd92dd973f4696bd03d919a98db0c9ab3/src/clic.adoc?plain=1#L1029

I'd suggest reducing this line just to "if (rs1 != x0)"

jb-brelot-nxp, 2024-09-23:
need to be updated in two places.

WIl lcare about this week.

christian-herber-nxp, 2025-04-07:
feature was removed

closed 2025-04-07 by christian-herber-nxp