riscv/riscv-fast-interrupt #420: Recommend moving definition of indirect CSRs into their own chapter and clarifying they are accessed via M-mode or S-mode CSRs.

james-ball-qualcomm, 2024-09-30:
'The CLIC registers accessed via the indirect CSR mechanism don't belong to a particular mode (M or S). Instead, they exist outside the scope of any particular mode but are accessed via different indirect CSR addresses for M-mode or S-mode. The only difference is that M-mode access sees all the information in the indirect access CSRs where as S-mode "any interrupt i that is not accessible to S-mode appears as hard-wired zeros in clicintip[i], clicintie[i], clicintattr[i], and clicintctl[i]."

We should move the definition of these indirect access CSRs back into their own chapter and then just have smclic and ssclic chapters talk about how to access them from their respective modes and how S-mode sees a subset of the info (the quoted text in the previous chapter)?

james-ball-qualcomm, 2024-10-03:
Do you all want me to make this change to the CLIC doc?

christian-herber-nxp, 2024-10-04:
I think it is reasonable given the current organization.
However, your suggestion made me realize that clicintattr.mode probably should not have the same behavior.
This is a separate issue and i will open that for discussion.
From my point of view, you can go ahead with this one.

christian-herber-nxp, 2025-05-05:
current state looks fine.

closed 2025-05-05 by christian-herber-nxp