riscv/riscv-fast-interrupt #424: clicintattr.mode should not be writable from S-mode

christian-herber-nxp, 2024-10-04:
Looks like a security hole to me. If M-mode assign an interrupt to S-mode, it does not sound ok for S-mode to throw that back. Seems like an easy way to corrupt M-mode.

I would suggest adding something along the lines of:

> When in mode x, it is only allowed to write values to clicintattr.mode which correspond to equally or less privileged modes. Writing other values is reserved.

(Please help me with the language)

christian-herber-nxp, 2024-10-07:
I overlooked the following note:

> For security purpose, the mode field can only be set to a privilege mode that is equal to or lower than the currently running privilege mode and if interrupts are supported at that privilege mode (e.g. ssclic extension).

This should not be a note, but part of the normative spec.

smatherd, 2025-04-08:
can we close this now that we are using deleg instead of clicintattr to control priv?

closed 2025-04-08 by christian-herber-nxp