riscv/riscv-fast-interrupt #426: In Appendix, we should not have CLIC interrupt ID recommendation chapter.

jb-brelot-nxp, 2024-10-07:
Appendix F: CLIC Interrupt ID ordering recommendations is indicating how we can potentially compress the vector table.by using another mapping,

We are recommanding to follow just the first mapping :

> ID  Interrupt   Note
> 
>  0  reserved
>  1  ssip        Supervisor software Interrupt
> 2  reserved
>  3  msip        Machine software interrupt
> 
>  4  reserved
>  5  stip        Supervisor timer interrupt
>  6  reserved
>  7  mtip        Machine timer interrupt
> 
>  8  reserved
>  9  seip        Supervisor external (PLIC/APLIC) interrupt
> 10  reserved
> 11  meip        Machine external (PLIC/APLIC) interrupt
> 
> 12  reserved
> 13  reserved
> 14  reserved
> 15  reserved
> 
> 16  csip        CLIC software interrupt
> 17+ inputs      CLIC local inputs

Silabs-ArjanB, 2025-03-10:
Hi @jb-brelot-nxp appendix F was largely the result of https://github.com/riscv/riscv-fast-interrupt/issues/89. Please note that it is key that that appendix is 'just a recommendation', not a part of the actual (mandatory part of the) specification.

christian-herber-nxp, 2025-04-07:
its gone

closed 2025-04-07 by christian-herber-nxp