riscv/riscv-fast-interrupt #428: Debug entry during hardware vectoring.

MarkHillCodasip, 2024-10-14:
Specification says that:

>  In CLIC mode, the dpc CSR additionally may hold the faulting address if breakpoints are allowed to trap on the table fetch during hardware vectoring. If breakpoints are allowed to trap on the table read, dret should honor xinhv.

However, I don't think xinhv can be used for this purpose because as well as being set during the implicit vector table fetch  it will also be set if debug mode is entered while in the middle of executing a fault handler on a failed vector table fetch. In this situation xinhv will be set but dpc will not be pointing to a trap vector table entry.

If debug mode can be entered with dpc pointing to a trap vector table entry I think the debug state needs its own inhv bit to indicate that.

MarkHillCodasip, 2024-11-25:
In addition, this section should clarify the behaviour when single stepping, in particular, unless a debug inhv bit is added to the architecture, when stepping through an interrupt vectoring operation it is not possible to stop between the return and the vector table fetch. However, if the vector table fetch traps this violates the  Debug Spec's definition of a single step:  "execute a single instruction **or** trap and then re-enter Debug Mode"

smatherd, 2025-04-28:
We discussed in TG meeting 4/14/25 that Zcmt has the same issue.  So it probably makes more sense to remove mention in CLIC spec and give the task to debug group to settle in a common way.  Pull #513 removes inhv text.

christian-herber-nxp, 2025-05-07:
inhv removed https://github.com/riscv/riscv-fast-interrupt/pull/513

closed 2025-05-07 by christian-herber-nxp