riscv/riscv-fast-interrupt #432: Clarify which CLIC CSRs are accessible in non-CLIC mode

mark-honman, 2024-11-13:
Section 3.8 of the spec "CLIC CSRs" states:
Unless explicitly specified differently below, when not in CLIC mode, behavior of CLIC-only CSRs should be same as if CLIC was not implemented, so should be treated as a "reserved" operation. 

I could not find any CLIC CSR that is stated to be accessible in non-CLIC mode. It would be helpful for the table in section 3.8 to specify the CLIC CSRs that are not treated as "reserved" in non-CLIC mode (if there are any) and for the statement quoted above to be specific, i.e. "When not in CLIC mode, behavior of CSRs introduced by this specification should be same as if CLIC was not implemented, so should be treated as a "reserved" operation. Exceptions are called out in the following table."

smatherd, 2025-04-07:
with #464, this text is removed.  ok to close?

mark-honman, 2025-04-07:
Yes! New spec is _so_ much cleaner!