riscv/riscv-fast-interrupt #433: MNXTI CSR clarifications?

mark-honman, 2024-11-14:
Section 3.8.7 "Next Interrupt Handler Address and Interrupt-Enable CSR"

It seems like the use of CSRRS was added after the text was written, leading to some inconsistencies. 

- text refers to "mstatus CSR is used for the RMW portion of the operation" but the CSRRS example only affects the bottom 5 bits of mstatus. 
- text refers to "must have a level greater than the saved interrupt level mcause.mpil" but the CSRRS example does not check this value, and instead uses rs1[23:16]. Maybe this is intentional but there is no explanation...
- It seems weird that CSRRS is permitted but not CSRRC.
- Similarly weird, the CSRRSI immediate cannot include non-zero immediate values for bits 0, 2, and 4 but no such restriction applies to CSRRS or CSRRCI.

hirooih, 2024-12-19:
> - text refers to "must have a level greater than the saved interrupt level mcause.mpil" but the CSRRS example does not check this value, and instead uses rs1[23:16]. Maybe this is intentional but there is no explanation...

I also would like to know the use-case of `csrrs rd, xnxti, rs1 (rs1 != x0)`.  All of sample codes in Appendix use only `csrrsi/csrrci xnxti`.

The use-case for `csrrs rd, xnxti, x0` is described:

> This can be used to determine if an interrupt could be taken without actually updating mil and exccode.

But in this case `mcause.mpil` should be used instead of `x0[23:16]`.  Pseudo-code for this is very simple and easy to implement by reusing logic for `csrrsi/csrrci xnxti`.

```cpp
 if (clic.priv==M && clic.level > mcause.mpil && clic.level > mintthresh.th) {
   rd = VTBASE + XLEN/8 * clic.id; // Return pointer to trap handler entry.
 } else {
   rd = 0;
 }
```

The current specification of `csrrs rd, xnxti, rs1 (rs1 may be x0)` has no known use-case and is not easy to implement.

christian-herber-nxp, 2025-04-07:
nxti was removed

closed 2025-04-07 by christian-herber-nxp