riscv/riscv-fast-interrupt #442: Wavedrom-style register definitions do not render in clic.adoc

mark-honman, 2025-01-08:
Since the new wavedrom style expression of the register layouts are not rendered when viewing clic.adoc on github, the only way to figure out the register layouts is to try to understand what's in the .edn files. 

I found that pasting the contents of the .edn files into wavedrom (online editor or Windows version) did not help because these versions of wavedrom could not interpret the .edn code.

christian-herber-nxp, 2025-01-08:
i see only two alternatives.
1) building locally
2) the document is being build for any PR (and any change to that). E.g. this late one: https://github.com/riscv/riscv-fast-interrupt/actions/runs/12637230786, you can download the build artifacts which contains the pdf.

(2) is not great but easier than (1) - or option (0) which is reading the source.

xiaoweish, 2025-01-09:
Try this wavedrom extension for chrome, it works for .adoc but not for .edn.

https://github.com/wavedrom/github/tree/master

Tests:
https://github.com/riscv/riscv-isa-manual/blob/814511cf02507bf30f5745c075243a3c0e037454/src/images/wavedrom/menvcfgreg.adoc

https://github.com/riscv/riscv-isa-manual/blob/main/src/images/wavedrom/menvcfgreg.edn

mark-honman, 2025-01-09:
Thanks for the tips - using the most recent build artifacts did the job for me.

closed 2025-01-09 by mark-honman